WebControlCNC/WebControl #141: Discussion: Teams

madgrizzle, 2020-05-19:
It seems that the best way to manage access is to establish and use only teams (versus individual invites to specifc repos).  I've not used teams before in github, so am looking for suggestions.  I don't think its a good idea to just make one team and make everyone a member of it with near-admin rights.  So... thoughts?

emilecantin, 2020-05-19:
I think an approach where we create teams for specific roles would make sense. E.g. "Developers", and "Triagers", the first having write access to basically everything (but they can't edit permissions), and the latter having the permissions to manage issues (triaging issues is a good way for less technical people to help out on an opensource project). If possible, I'd also restrict pushing to the master branch & tagging releases to a specific team as well ("Maintainers", perhaps?).

madgrizzle, 2020-05-19:
I created three teams.. first responders, developers, and maintainers.  I added you, @emilecantin and @gb0101010101, as members of the @WebControlCNC/maintainers team.  I'm not sure the process someone has to go through to request a joining a team.

gb0101010101, 2020-05-19:
> I've not used teams before in github, so am looking for suggestions.

TBH I have not setup teams on Github either. You might want to chat with @BarbourSmith who is the owner of https://github.com/orgs/MaslowCNC

Thanks for adding me to team.

madgrizzle, 2020-05-19:
I don't think bar setup any teams in the maslow org.

madgrizzle, 2020-05-20:
I created a test github account and it doesn't appear to be any way to request an invite to join the organization.  You must be part of the organization before you are added to a team.  So we really need some kind of welcome page that tells people that if they want to join they have to be invited.  There might be a proper way of doing it, but I haven't found it yet.  As an option, we can create an issue that's pinned (I pinned this one) where people can post requesting an invite and what, if any, team they desire to be a member of.  I'm thinking to try to control things, they should also be a member of the maslowcnc forum as well... so they'd have to provide both github name and forum name.

Also, I created a users team for people that just want to join, but not necessarily can help in any manner.

gb0101010101, 2020-05-22:
> I created a test github account and it doesn't appear to be any way to request an invite to join the organization.

Yep I tested the same. If the repo is public then the average person will not need to join organization to fork code and submit PR. Only people that get more permission need to join.