WebControlCNC/WebControl #142: Join the Organization!

madgrizzle, 2020-05-20:
There's no requirement to join the WebControl CNC organization, but if you would like to help out please feel free to do so.  There's no method (that I've yet discovered) in github to allow someone to request joining the org, so you have to be invited.  To join, just post here with the following:

1.  Github username
2.  MaslowCNC forum username (https://forums.maslowcnc.com/)  *Must be part of this forum to join*
3.  When did you get your Maslow and from whom?
3.  Which team you want to join.

We have the following teams created:

1. Users - Anybody that wants to join and just follow what's going on.  Open to all
2. First Responders - People that want to help and solve user issues that aren't necessarily software issues and if their are software issues, let the developers know. 
3. Developers - People that want to create new features, and fix bugs, do cool things, etc. 
4. Maintainers - People that want to manage the repo and create releases (keeping this group small for what should be obvious reasons)

thormj, 2020-07-13:
So, I'm thormj, and I got the KS Maslow and am running it with the metal pantograph, ryobi plunge router, and z-axis.
Put me in for the Users and Developers (WebControl / Firmware) groups.

I noticed that PyCharm has a freebie for free software groups (I'd let the maintaners handle the keys)... just FWIW since it took be about 20 minutes to run down an errant parenthesis in Javascript...

rolambert, 2020-07-15:
Hello all,

+ rolambert
+ rolambert
+ purchase the M2 on pre-order waiting for it to ship
+ dev team please

> the project looks good, cant wait to play around with it some more. my back ground is .py and .net

Orob-Maslow, 2020-07-15:
You won't be using webcontrol with m2.

rolambert, 2020-07-15:
> You won't be using webcontrol with m2.

what open source software is compatible with the m2, that I 'will' be using?

Orob-Maslow, 2020-07-15:
Cncjs.  Look in the makermadecnc GitHub repo

gb0101010101, 2020-07-16:
Hey @thormj & @rolambert, tried to add you as members but do not have permission. Looking into getting this fixed.

gb0101010101, 2020-07-16:
@thormj & @rolambert, Invites sent.

rolambert, 2020-07-16:
> @thormj & @rolambert, Invites sent.

Cheers!

Orob-Maslow, 2020-07-16:
Sorry, not trying to be rude.  The Arduino due and due shield are upgrades
to the Arduino mega and accompany the full M2 kit and run a full version of
grbl compliant firmware that doesn't work with webcontrol.  The goal/
intent was for maslow to get there with the mega and webcontrol.  It
eventually will.

GeroBH, 2020-08-04:
1. Github username: GeroBH
2. MaslowCNC forum username: Gero
3. When did you get your Maslow and from whom?: I got the kit Apr 2017 as part of the kickstarter campaign from Hannah/Bar.
4. Which team you want to join?: Users first. First Responders later (after more experience).