WebControlCNC/WebControl #146: WebControl does not work behind HTTPS due to hardcoding of HTTP schema

zaneclaes, 2020-05-21:
It appears the `socket.io` interface config may have hardcoded the HTTP schema into the URL construction, because WebControl fails to work behind a HTTPS connection for me:

[image]

madgrizzle, 2020-05-21:
Yes, it is.  HTTPS was another layer of things that I never tried to deal with considering my relatively limited knowledge of such things.  static/scripts/baseSocket.js has it set to open via HTTP from the browser side.  I don't know what specifically needs to be done to support operation over HTTPS.  I "assume" webcontrol would need to be configured for HTTPS.. however that's done.

zaneclaes, 2020-05-21:
All you need to do to adapt to schema is leave out the schema from any client side code. Assets on the browser side are resolved relative to the root schema when no schema is provided. For example, the following HTML tag:
`<link rel=dns-prefetch href=//fonts.googleapis.com>`

Will resolve to http on a HTTP site and HTTPS on a HTTPS site. Nothing else changes, as long as the backend implements TLS correctly (not in your control).

Same applies to JavaScript.

madgrizzle, 2020-05-21:
It seems to still run if I drop the HTTP out of the baseSocket.js file, but webcontrol itself is running the flask development server and its not running HTTPS (not everything is done via websocket).  Does that still cause an issue?

zaneclaes, 2020-05-21:
tl;dr: that sounds fine...

I'm not sure what you mean by:
>  its not running HTTPS

But I suspect what you mean is that your flask server does not implement TLS termination (i.e., negotiate a HTTPS handshake with the client). Indeed, that is the job of the front-proxy. Right now, you're doing this:

`localhost:5000` -> `flask:5000`

Nginx is a common front proxy which can implement TLS termination. Its job is to relay the traffic to the Flask server, having handled the encryption/decryption and passing off normal HTTP traffic:

`https://house.snowy-cabin.com` -> `nginx:443` -> `flask:5000`

madgrizzle, 2020-05-21:
You're correct in your suspicion.  I've used nginx on an aspnet program, but that's on a dedicated webserver and not sure how to (or if its possible) package it all up into a single pyinstaller executable.  That's sort of the tricky part.

zaneclaes, 2020-05-21:
Oh, you don't need to package nginx. Just make the change to the frontend that you suggested (and test that it's doesn't break your deployment). It's the end-user's responsibility to run the front proxy.

For example, I have WebControl running in a container right now. I can access it directly at `192.168.0.202:5000` (connecting to flask). In my original screenshot, you can see that I am also able to load the UI through the HTTPS website URL `https://maslow.snowy-cabin.com`. The reason this works is that my front proxy receives the traffic to that URL, and acts, well, as a proxy! So the WebControl app doesn't need to know anything about the fact that it's being accessed over HTTPS. From its perspective, it's just getting normal old traffic via port 5000.

The change is required because the frontend has to know which URL to connect to. As long as it does not respect the HTTPS schema, the request fails to go through the front poxy **even though** I loaded the website through the front proxy. This causes browser security errors, as you can see from the Javascript log.

madgrizzle, 2020-05-21:
Perhaps we are talking about two different kinds of installations.  For the releases I build, they function as all-in-one webservers (for lack of a better term).  It's basically a single installation containing everything the user needs.  They don't have to set up a webserver/frontend, etc.

zaneclaes, 2020-05-21:
I agree. That's what I'm describing as well.

I think the part you're missing is that a front proxy like `nginx` is an entirely optional add-on which **should** be left in the user's control. This is how server deployment is **supposed** to be done, and also in keeping with containerization best-practices. Think of `WebControl` like, say, `WordPress`:

Wordpress:
* Is a PHP server application
* Is available as a docker container
* Has a frontend (HTML + Javascript) that can connect via sockets to the backend (PHP)

WebControl:
* Is a Python server application
* Is available as a docker container
* Has a frontend (HTML + Javascript) that can connect via sockets to the backend (Python)

Wordpress does not include nginx is you installs in on your computer, no should it. Nor does it support SSL/HTTPS out of the box. If you want to host your WordPress site anywhere but at `localhost:xxx`, then you add your own nginx (or other front proxy) to achieve load balancing, SSL termination, etc.

But most importantly, **none of this changes the way the app works when nginx is not around**. Wordpress and WebControl both function as normal, without any awareness of nginx. I can still load Wordpress and run it on `localhost:xxx`, and in fact that is the default behavior unless I decide to add nginx. Nor should they have any awareness of nginx, because it is not their concern. To try to couple the two would be to re-invent the wheel and introduce a whole hell of a lot of pain.

Finally, the real point here is not about the backend. It's about the frontend. **WebControl** has a **security vulnerability** because of the way its frontend is designed. In a nutshell: 

> WebControl's frontend is attempting to connect to a different server than where the backend resides. It does not use the same schema as it was told to, and this lack of following of security practices is the root cause of the problem.

This is a classic security vulnerability which leaves WebControl open to man-in-the-middle attacks, ultimately allowing an attacker to theoretically take control over the Maslow (literally, control it) in certain scenarios where the WebControl application is not otherwise secured.

Orob-Maslow, 2020-05-21:
Side question... Does haproxy do the same thing that nginx is doing?  Octoprint uses haproxy in a manner similar to your description.

emilecantin, 2020-05-21:
Yes, nginx is a web server that happens to be often used as a reverse
proxy, while haproxy is purpose-built for that.

zaneclaes, 2020-05-21:
Which is another reason I was recommending against bundling any sort of
reverse proxy (or the assumption of its existence) with the application
itself :) I happen to use Envoy Proxy myself (nginx is merely a convenient
example) so my setup will look a touch different.

madgrizzle, 2020-05-22:
I can drop the http from baseSocket.js if that works for your installation because it *seems* unneeded for a standard install (will check more).  I just want to make sure there are no other changes needed.

zaneclaes, 2020-05-22:
Sounds great — if you want, you can just tell me where the config file is (or open an PR to show me) and I can test it myself in the docker container as well.

madgrizzle, 2020-05-22:
PR committed into master branch.

closed 2020-05-22 by madgrizzle