WebControlCNC/WebControl #156: Discussion: Vision for the Future of WebControl

madgrizzle, 2020-05-28:
@WebControlCNC/maintainers , @WebControlCNC/developers 
I'd like to have a discussion on the "vision for the future of webcontrol" but rather it seems that doing so in the "issues" isn't really appropriate.  Is there a better way to do so?  It's not just for the application, but also the firmware, so its more of a WebControlCNC kind of discussion.

closed 2020-05-28 by madgrizzle

davidelang, 2020-05-30:
do it on the forum

David Lang

On Thu, 28 May 2020, madgrizzle wrote:

> @WebControlCNC/maintainers , @WebControlCNC/developers
> I'd like to have a discussion on the "vision for the future of webcontrol" but rather it seems that doing so in the "issues" isn't really [...]

madgrizzle, 2020-05-30:
I created a discussion in the teams section where there's been a bunch of discussion.  Ive invited you a few times to join the org, but it's still pending.  I tried the email address again, so hopefully that will work.

zaneclaes, 2020-05-30:
Relevant:
https://forums.maslowcnc.com/t/documenting-maslow-what-am-i-missing/13537/2