WebControlCNC/WebControl #159: gcode stops running randomly; last logs are "5.08,WPos:0.000,0.000,0.000>"

zaneclaes, 2020-05-29:
I originally configured my Maslow with GroundControl and then imported my settings to WebControl. I've tried two cuts with WebControl, both of which stopped within the first few minutes. By "stopped," I mean that the all motors ceased operation and the sled has now remained still for over 10 minutes. 

The last logs look like this:

```
Sending: G1 X-22.4770 Y-16.15840  
Sending: G1 X-22.3940000000000019 Y-16.23840  
Sending: G1 X-22.3120 Y-16.31910

5.08,WPos:0.000,0.000,0.000>
5.08,WPos:0.000,0.000,0.000>
5.08,WPos:0.000,0.000,0.000>
5.08,WPos:0.000,0.000,0.000>
```

@madgrizzle any thoughts?

madgrizzle, 2020-05-29:
The logs seem weird because its cutoff.  Are you able to go to Actions->Download Diagnostircs File and send to madgrizzlemaslow at gmail dot com?

madgrizzle, 2020-05-29:
and email the gcode as well?

zaneclaes, 2020-05-29:
I'm going to try swapping it out for a RPi4 instead of RPi3B+, which has a much better CPU. Here's the CPU usage monitored by Prometheus on a RPi3B+:
[image]

madgrizzle, 2020-05-29:
I'm concerned about the cut-off logs.. the last lines with 5.08, WPos:0.00...> should look like <MPos: xxx.xx, xxx..xx, 5.08, WPos:0.00 ...>

I wonder if there was a serial hiccup and part of webcontrol crashed because of malformed log data.

madgrizzle, 2020-05-29:
Reset your chains (sled might have lost position if there was a usb-disconnect) and then skip ahead using the > buttons (or use the goto) to the line where it stopped working and hit play.  If it works as intended, you should be able to pickup from where it left off.

zaneclaes, 2020-05-29:
> I wonder if there was a serial hiccup and part of webcontrol crashed because of malformed log data.

That seems plausible. I hope it's not hardware related; I have the USB cable running along the ceiling to the Maslow from the control unit.

> I would be hesitant about running the gui/browser concurrent with webcontrol on an RPI 3B+. Not sure what load the camera would put on. Are you using the camera feature of webcontrol or something else.. because webcontrol's camera feature is taxing as well.

Fair enough. I did try letting the display go blank (less CPU) and turning the camera off.

zaneclaes, 2020-05-29:
Okay, definitely a CPU issue. RPi4 fixed this bigly. ~33% CPU usage now. This is my first home-grown gcode cut (used Easel to convert SVG).
[image]

zaneclaes, 2020-05-29:
Shoot... I have to re-open this. Looks like the Pi4 delayed the onset of the problem, but did not fix it. It takes ~1 hour of cutting to get to a disconnect, now. I actually think the whole container may be getting killed. The server becomes unresponsive for a couple minutes and then then gcode is not loaded any more. It may also be a networking issue, given that my garage is on the border of wifi range. I'm trying to narrow down symptoms now...

madgrizzle, 2020-05-29:
Can you send me the gcode.. to make sure it isn't failing on something in it?

madgrizzle, 2020-05-29:
I looked at the gcode and it seems tame (easel gcode is what I used when I first started working on webcontrol).  I tend to think something is hiccuping the serial connection.  Either there's a RPI issue with a lot of things running (possibly camera) or there's a USB connection issue.  If data coming from the arduino gets corrupted, its possible that some of the parsing fails and causes one of the threads to crash, but I try to catch those situations so I don't really know.  If you can email me your log.txt file from the .WebControl folder, that might tell us what caused it to crash.

zaneclaes, 2020-05-29:
Sent! On the most recent stop, I actually did get a dialog that the USB had disconnected. And I haven't seen the Docker container actually definitively crash, so it may be that...

madgrizzle, 2020-05-29:
I'll look at log later tonight or in the morning.  How long of a USB cable is between the rpi and the Arduino?

zaneclaes, 2020-05-29:
It's the long blue cable that comes with the Maslow Classic kit. I want to say ~15ft.

zaneclaes, 2020-05-29:
Just briefly looking at `alog.txt`, this suspicious block seems to appear before many of the restarts. I don't believe there's any actual chain-over-sprocket error...
```
19.49: $32=1.00000000 (z axis Pos proportional weight)
19.51: $33=5.00000000 (z axis Kp Velocity)
19.52: $34=0.00000000 (z axis Ki Velocity)
19.54: $35=0.27999999 (z axis Kd Velocity)
19.56: $36=1.00000000 (z axis Velocity proportional weight)
19.58: $37=81.73586273 (chain sag correction value)
19.64: $38=2 (chain over sprocket)
19.66: $39=3 (PWM frequency value 1=39,000Hz, 2=4,100Hz, 3=490Hz)
19.68: $40=0.00000000 (chain tolerance, left chain, mm)
19.74: $41=0.00000000 (chain tolerance, right chain, mm)
19.76: $42=2.00000000 (position error alarm limit, mm)
19.77: ok
```

madgrizzle, 2020-05-29:
That looks like just the settings being reported by the controller.. this happens after a restart so webcontrol can sync up the controller and send any setting updates that are needed.

madgrizzle, 2020-05-30:
15 foot cable might the source of the problem.  Some people have had issues with even shorter cables.

madgrizzle, 2020-05-30:
If you sent the log, I didn't receive it.

Orob-Maslow, 2020-05-30:
@zaneclaes I use a 1' cable on a headless rpi3 (with external start/stop buttons) and the once I had a problem similar to your description when I turned on the "buffer gcode" option in webcontrol settings.  It defaults to "off," so it likely is not your issue, but you might look at that setting to verify.

zaneclaes, 2020-05-30:
> If you sent the log, I didn't receive it.
Sorry, got stuck in the upload; one of them was 3Mb.

> I turned on the "buffer gcode" option in webcontrol settings. It defaults to "off," so it likely is not your issue, but you might look at that setting to verify.
Thanks for the tip; confirmed that it's off.

I'd prefer to drive the camera with the same RPi that drives the Maslow, for space saving purposes. However, I do have a super-long camera cable, so perhaps I can move the Pi to the back of the frame and switch to a shorter USB...

zaneclaes, 2020-05-30:
In further debugging, I've found a few bugs...

WebControl always reports CPU usage of 10-40% range. However, Kubernetes is able to inspect the container usage with greater precision, and finds that this value is inaccurate. This would not be surprising if WebControl had _overestimated_ the usage, because it may have been constrained to the containers' definition of maximum CPU/RAM, but the fact that it's _underestimating_ usage suggests a bug:
[image]

Note that the CPU dropped precipitously after I turned the camera off near the end. This is expected of course, but what caught my attention is the network utilization. WebControl was sending ~10Mbps and receiving ~5Mbps. That's more than Netflix. 

As an experiment, I turned off the camera and instead loaded up MotionEye (a piece of software meant for turning RPis into webcams). I set it to stream 1080p resolution at 30FPS and let it run in one browser, and WebControl (with the camera off) in another browser. Note that WebControl **still** uses more CPU than MotionEye. Also note that the network utilization is down to ~30 Kbps (roughly 0.3% of what WebControl required to stream the video):
[image]
The CPU spike happens when I load the WebControl UI via a browser, so this further points at massive inefficiencies in the socketIO connection. 

Finally, there appears to be a memory leak. We should not see these step increases in the memory utilization of the container. They seem coupled to usage of the web interface (note how it was constant at night, before I woke up):
[image]

If there are memory leaks within the socket IO (easy to do), WebControl will inevitably crash on machines like this (RPi4 with only 1GB RAM).

zaneclaes, 2020-06-02:
Okay, I've now cut 1.5 sheets of plywood using MotionEye for the camera without any crashes. I think it's safe to close this now.

If we're going to keep investing in WebControl over CNCjs, we should address these issues lest other users stumble upon them. @emilecantin just FYI I have easy ways to benchmark my cluster like in the above screenshots (if you want me to do so for CNCjs).

closed 2020-06-02 by zaneclaes