WebControlCNC/WebControl #163: Discussion: Integration of HackMD.io

madgrizzle, 2020-06-03:
I linked some more files.. it takes time to go through the process of adding a file.  I also put in a tagging scheme that might work for us.  The "webcontrol" tag currently identifies the document as part of the online/offline help system.  The second tag, for example "Getting Started" identifies the folder for which the file resides within the docs directory.  If its in a subfolder of a folder, then I hyphenated/concatenated the directory names together... "Actions-Controller" as an example of a file that resides in the docs/actions/controller folder.  This seems to keep everything grouped together and I think its manageable.  Ideally there would be a hierarchical view, but this is what I came up with.  @zaneclaes, does this make sense or is there a better way?

zaneclaes, 2020-06-03:
Your use of tags makes sense to me. I wish this were a little easier to navigate; definitely not the silver bullet one may hope for.

With Bar's permission, I'd like to use the docs being created in HackMD by the forum participants for MaslowCNC.com ...

zaneclaes, 2020-06-04:
Update: Bar is happy to have us make updates to MaslowCNC.com. I've asked him how it's currently implemented (HTML? Markdown?) so we can incorporate it.

zaneclaes, 2020-06-05:
Couple quick questions @madgrizzle as I am organizing things...

- Is the "Wiki" page defunct? Is look like it was an abandoned replica of the `docs/` folder. If so, I cannot delete the pages; it may be best to redirect to the `docs`.
- What is the purpose of `Release: >0.906 {: .label .label-blue }` and the table header seen in many of the `docs/` markdown files? I don't recognize this syntax, and am not sure what purpose it is serving.

madgrizzle, 2020-06-05:
Yeah. Wiki was first attempt.. then moved to GitHub pages.  The fancy label is part of the formatting used by the GitHub pages Jekyll theme, justthedocs 

https://pmarsceill.github.io/just-the-docs/

madgrizzle, 2020-06-05:
I think I have all the doc files linked.  The assets are still using a relative reference.