WebControlCNC/WebControl #172: ISSUE: main.py   "from werkzeug import secure_filename ImportError"

rolambert, 2020-07-16:
main.py has a  '''dependancy''' issue:
 
+ I'm using: Werkzeug  -version 1.0.1  
+ could not get -version 0.15.3

Resolution, change main.py line 20 too:

```
from werkzeug.util import sercure_filename
```

gb0101010101, 2020-07-17:
Just checked and Werkzeug still has all the old version available 
```
# pip install Werkzeug==
ERROR: Could not find a version that satisfies the requirement Werkzeug== (from versions: 0.1, 0.2, 0.3, 0.3.1, 0.4, 0.4.1, 0.5, 0.5.1, 0.6, 0.6.1, 0.6.2, 0.7, 0.7.1, 0.7.2, 0.8, 0.8.1, 0.8.2, 0.8.3, 0.9, 0.9.1, 0.9.2, 0.9.3, 0.9.4, 0.9.5, 0.9.6, 0.10, 0.10.1, 0.10.2, 0.10.4, 0.11, 0.11.1, 0.11.2, 0.11.3, 0.11.4, 0.11.5, 0.11.6, 0.11.7, 0.11.8, 0.11.9, 0.11.10, 0.11.11, 0.11.12, 0.11.13, 0.11.14, 0.11.15, 0.12, 0.12.1, 0.12.2, 0.13, 0.14, 0.14.1, 0.15.0, 0.15.1, 0.15.2, 0.15.3, 0.15.4, 0.15.5, 0.15.6, 0.16.0, 0.16.1, 1.0.0rc1, 1.0.0, 1.0.1)
```
Did you use `pip install -r requirements.txt` to install it?

rolambert, 2020-07-20:
Aclean up solved it for me:

```
conda clean --all --yes
```

no issues installing it in a new env after than. cheeers!

closed 2020-07-20 by rolambert