WebControlCNC/WebControl #177: Lower case G-commands not visualized on screnn, the g-code is loaded and runs

GeroBH, 2020-08-06:
Getting Webcontrol to run on my Ubuntu (fake servo to play, research and understand, I noticed a g0, g1 would no be shown on the screen. Only upper case G0,G1 is shown.
To replicate replace all G with g in a .nc file and load.
Checking the huge amount of sample g-code i gathered debugging user files, all I randomly checked have a capital G, so this might be not be a big issue. Just thought I would mention, because i also checked Camotics and bCNC and they display the cut, even with a lower case g.
I stumbled across this loading a file for a 'lift-desk' that I have created with FreeCAD ~ 2 years back, so perhaps v0.16?.
Checked recent FreeCAD files and they all have the capital G.

Kind regards, Gero

davidelang, 2020-08-06:
I have seen other CNC machines that were case sensitive like this, so while I 
agree it would be good to make sure we aren't, it's not uncommon to be.

David Lang

On Thu, 6 Aug 2020, Gero wrote:

> Getting Webcontrol to run on my Ubuntu (fake servo to play, research and understand, I noticed a g0, g1 would no be shown on the screen. Only upper case G0,G1 is shown.
> To replicate replace all G [...]

Orob-Maslow, 2020-08-06:
I thought I saw in the webcontrol code that it converts all the gcode to
capital letters.  It doesn't save it, but it should not matter after the
conversion.  What does it show when you view the lines of code in the code
viewer?

GeroBH, 2020-08-06:
It shows all lines of the g-code and it also runs it. It is just not displaying on the screen.

Orob-Maslow, 2020-08-06:
Are they upper case in the viewer?

GeroBH, 2020-08-06:
No, lower case. Accidental closed this...

GeroBH, 2020-08-06:
[image]

GeroBH, 2020-08-06:
Reopened... Sorry

Orob-Maslow, 2020-08-06:
I'll look into it later today.

GeroBH, 2020-08-06:
Webcontrol, single folder Version v0.94, Holey Firmware 51.27, Ubuntu 18.04.4 LTS bionic, Fake-Server mode.

madgrizzle, 2020-08-06:
Webcontrol evolves over time and some functions get reorganized.   I'm sure this goes back to the change to not modify the gcode file itself.  If someone can try, add an .upper() call to gcodeFile.py around line 619.. 

>fullString = (fullString.upper()+" ")

Orob-Maslow, 2020-08-06:
I was looking at gcodefile.py and thinking line 253 and others like it
could be

if (command == "G00") or (command == "g00")

But converting to upper case earlier makes more sense and would work the
same.

GeroBH, 2020-08-06:
> gcodeFile.py

I did a locate to find the file, but it was found in a folder from Dec 2018. What did I do wrong?

whoareyou4@dexter:~/Maslow/webcontrol/single_directory$ locate gcodeFile.py
/home/whoareyou4/Maslow/fw-test/WebControl-master/File/gcodeFile.py

madgrizzle, 2020-08-06:
> I did a locate to find the file, but it was found in a folder from Dec 2018. What did I do wrong?

I'm referencing the gcodeFile.py file that's in the master branch on this repo.

Orob-Maslow, 2020-08-08:
This uses the code fix Madgrizzle suggested.  I tested it and also removed a file that got added a couple months ago as well.

https://github.com/WebControlCNC/WebControl/pull/178

closed 2020-08-08 by Orob-Maslow