WebControlCNC/WebControl #183: Webcontrol on Raspberry 400, running Ubuntu

bsailer2510, 2020-11-27:
I’ve installed Webcontrol on my brandnew Raspberry 400, running Ubuntu 20.10 (64bit). Version is
v0.920.

When i try to start the service manually i get the following error:

```
bs@bs-werkstaette:~/webcontrol$ ./webcontrol
bash: ./webcontrol: cannot execute binary file: Exec format error
```

The status request gives me:

```
    bs@bs-werkstaette:~/webcontrol$ sudo systemctl status webcontrol
    ● webcontrol.service - WebControl
         Loaded: loaded (/etc/systemd/system/webcontrol.service; enabled; vendor preset: enabled)
         Active: failed (Result: exit-code) since Fri 2020-11-27 09:27:12 CET; 35min ago
        Process: 761 ExecStart=/home/bs/webcontrol/webcontrol (code=exited, status=217/USER)
       Main PID: 761 (code=exited, status=217/USER)

    Nov 27 09:27:12 bs-werkstaette systemd[1]: webcontrol.service: Scheduled restart job, restart counter is at 5.
    Nov 27 09:27:12 bs-werkstaette systemd[1]: Stopped WebControl.
    Nov 27 09:27:12 bs-werkstaette systemd[1]: webcontrol.service: Start request repeated too quickly.
    Nov 27 09:27:12 bs-werkstaette systemd[1]: webcontrol.service: Failed with result 'exit-code'.
    Nov 27 09:27:12 bs-werkstaette systemd[1]: Failed to start WebControl.

```
I’ve no idea whats going wrong and the service want start. I’ve tried it as weel as su User.
Kindly ask to help and troubleshot.

gb0101010101, 2020-12-09:
```
cannot execute binary file: Exec format error
```
Did you download https://github.com/WebControlCNC/WebControl/releases/download/v0.920/webcontrol-0.920-rpi-singledirectory.tar.gz ? If not then start over using this file.

If you did use webcontrol-0.920-rpi-singledirectory.tar.gz.

Go to your "webcontrol" directory and enter `file webcontrol` and hit enter.
If it says "webcontrol: ELF 32-bit LSB executable, ARM...." then there is a problem running the 32bit binary on your 64bit OS.

Could be the ARM7 binary does not work on ARM8 processor, Ubuntu OS does not support 32bit on 64bit, the PI 400 processor does not support 32bit, or you are missing 32bit support in OS. I don't know off hand how to fix these issue on ARM Ubuntu.

You should still be able to run webcontrol from source code but it will be more involved:

1. On command prompt enter `pip3 --version` and hit enter.
2. If it does not report a version number e.g. "pip 9.0.1 ..." then `sudo apt install -y python-pip3` and hit enter.
3. Remove the previous webcontrol install then run the following to download and install source code.

```
cd ~/
wget https://github.com/WebControlCNC/WebControl/archive/master.zip
unzip master.zip
mv WebControl-master webcontrol
cd webcontrol
pip3 install -r requirements.txt
```

You can then run webcontrol using `python ~/webcontrol/main.py`

gb0101010101, 2021-12-03:
Closing issue due to no response.

closed 2021-12-03 by gb0101010101