WebControlCNC/WebControl #184: Getting (code=exited, status=203/EXEC) when trying to start the WebControl service

B0RGN1NE, 2021-02-08:
Good Day!
I am attempting to set up a dedicated PC to run WebControl using a fresh install of Debian Linux 10 (buster) 32bit OS
getting the following message:

● webcontrol.service - WebControl
   Loaded: loaded (/etc/systemd/system/webcontrol.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-02-07 23:49:07 PST; 24min ago
  Process: 2176 ExecStart=/home/b0rgn1ne/webcontrol/webcontrol (code=exited, status=203/EXEC)
 Main PID: 2176 (code=exited, status=203/EXEC)

Feb 07 23:49:07 Maslow01 systemd[1]: webcontrol.service: Service RestartSec=100ms expired, scheduling restart.
Feb 07 23:49:07 Maslow01 systemd[1]: webcontrol.service: Scheduled restart job, restart counter is at 5.
Feb 07 23:49:07 Maslow01 systemd[1]: Stopped WebControl.
Feb 07 23:49:07 Maslow01 systemd[1]: webcontrol.service: Start request repeated too quickly.
Feb 07 23:49:07 Maslow01 systemd[1]: webcontrol.service: Failed with result 'exit-code'.
Feb 07 23:49:07 Maslow01 systemd[1]: Failed to start WebControl.

This is my webcontrol.service file:

[Unit]
Description=WebControl
After=network.target

[Service]
ExecStart=/home/b0rgn1ne/webcontrol/webcontrol
WorkingDirectory=/home/b0rgn1ne/webcontrol
StandardOutput=inherit
StandardError=inherit
Restart=always
User=b0rgn1ne

[Install]
WantedBy=multi-user.target

any suggestions?

gb0101010101, 2021-02-26:
Does your path "/home/b0rgn1ne/webcontrol/webcontrol"  lead to file that has execute permission?

gb0101010101, 2021-12-03:
Closing issue due to no response.

closed 2021-12-03 by gb0101010101