WebControlCNC/WebControl #189: WebControl taking down my network

buck2825, 2022-01-16:
I have a moderately complex home network.  I connect my RPi to wifi but at some point the pi starts issuing IP addresses.  My Pi-hole is what I want to issue my IP addresses.  I think this is caused by start up times of the different systems.  

should the pi hole not issue IP addresses if connected to an already existing wifi network.

davidelang, 2022-01-17:
when you say 'issuing IP addresses' are you talking about DHCP or DNS?

pi hole should be your DNS server, but not DHCP

David Lang

On Sun, 16 Jan 2022, buck2825 wrote:

> I have a moderately complex home network.  I connect my RPi to wifi but at some point the pi starts issuing IP addresses.  My Pi-hole is what I want to issue my IP addresses.  I think this is caused [...]

buck2825, 2022-01-17:
DHCP, I have 6 other devices on my network that can be DHCP servers, but I have them shut off or on as required.  the Maslow is the only one that turns on and off when it wants to.   based on network connectivity.

davidelang, 2022-01-17:
the maslow webcontrol image hs the ability to be it's own AP so that you can 
connect to it when you don't have any other network up, but it should not be 
enabled once you configure it to connect to an existing network.

This is a serious bug in the software, but I would say that the fix is to login 
to the pi and just uninstall the dhcp server (I don't know if it uses dhcpd or 
dnsmasq or something else). If you have an existing network defined it should 
not be used.

On Mon, 17 Jan 2022, buck2825 wrote:

> DHCP, I have 6 other devices on my network that can be DHCP servers, but I have them shut off or on as required.  the Maslow is the only one that turns on and off when it wants to.   based on network connectivity.

cstoker2, 2022-01-18:
My Maslow pi with webcontrol started doing the same thing about a month
ago.  I had to edit /etc/dnsmasq.conf and remove the dhcp server.   You can
also stop it with:
sudo service dnsmasq stop
but it will restart at boot if not disabled in the config file.

Super annoying that some update enabled this, because I sure as hell didn't.

buck2825, 2022-01-29:
I ended up removing dnsmasq so the RPi couldn't issue IP addresses
`~$ sudo apt --purge remove dnsmasq
 ~$ sudo apt autoremove`