WebControlCNC/WebControl #190: Webcontrol on a Raspi 4 - always failing... HELP!

LennyInCanada, 2022-02-06:
Good day good friends!
I am helping a friend to set up her Maslow with the Raspberry Pi 4. I am a newbie to Linux and have given myself a bit of a haircut by pulling out a bit my my remaining hair. The Webcontrol.service does not want to start and is giving errors - see below. I have tweaked the permissions, the file locations, and gone through following the GitHub steps for setting up, but something is still off. I have reached out to a couple of Linux Ninjas but no success there. Please see below and let me know if you have any suggestions. I am all eyes and ears, and a bit less hair!

 pi@webcontrolcnc:~ $ sudo systemctl status webcontrol
* webcontrol.service - WebControl
   Loaded: loaded (/etc/systemd/system/webcontrol.service; enabled; vendor prese
   _Active: failed (Result: exit-code) since Sun 2022-02-06 17:36:57 EST; 3min 54
 Main PID: 723 (code=exited, status=203/EXEC)_

Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Service RestartSec
Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Scheduled restart 
Feb 06 17:36:57 webcontrolcnc systemd[1]: Stopped WebControl.
Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Start request repe
Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Failed with result
**_Feb 06 17:36:57 webcontrolcnc systemd[1]: Failed to start WebControl._**
lines 1-11/11 (END)...skipping...
* webcontrol.service - WebControl
   Loaded: loaded (/etc/systemd/system/webcontrol.service; enabled; vendor preset: enabled)
   **_Active: failed (Result: exit-code) since Sun 2022-02-06 17:36:57 EST; 3min 54s ago
 Main PID: 723 (code=exited, status=203/EXEC)_**

Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Service RestartSec=100ms expired, scheduling restart.
Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Scheduled restart job, restart counter is at 5.
Feb 06 17:36:57 webcontrolcnc systemd[1]: Stopped WebControl.
Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Start request repeated too quickly.
Feb 06 17:36:57 webcontrolcnc systemd[1]: webcontrol.service: Failed with result 'exit-code'.
**_Feb 06 17:36:57 webcontrolcnc systemd[1]: Failed to start WebControl._**

Thank you for your assitance!
Best regards,
Lenny

rspeed, 2022-02-15:
What does `sudo journalctl -xe` show?

LennyInCanada, 2022-02-27:
Hi Rob,All good with the full reinstall of a Raspberry Pi with WebContol all built into the OS install. We got the Malsow CNC running yesterday. A bit of fine tuning required to as the x axis seems a bit off, making a circle more oval than round. Also the router has a bit of a wobble as it sits loosely in the sled housing.Thanks again!LeonardSent from my Galaxy

MStylesMS, 2025-01-17:
I know this thread is old, but hope one of you with insight is still listening and can give an answer off the top of your head.

My question is if two instances of WebControl can run on the same machine at the same time?  I have a spare Pi4, two Maslow's, and more than a couple of old monitors laying around.  I'm trying to get my shop back running and would like to drive both Maslow's from the same Pi4.  

I haven not dug into it yet, but I am guessing that each instance of WebControl is limited to operating a single machine at a time.  What I don't know is if two version could be running on the same Pi, each with a different config.  One for machine A with COM port A and address xxx.xxx.xxx.xxx:5000 and then B with COM B and address xxx.xxx.xxx.xxx:5001.  

If WebControl is running it's own DHCP that's a concern to me because I also have complex network (e.g. 192.168.0-15.xxx) so having competing DHCP's concerns me.  I was going to use mDNS to rename the machine to something like maslow:local, but that's all.

Any insights before I embark on possibly bringing down my network?  :-D

Orob-Maslow, 2025-01-17:
Tldr: You theoretically can. 

 The nuances are going to be serial device identification, which should be easy to distinguish and ethernet port usage.  The default is port5000 so ipaddress:5000.  You will need to change one of the instances to be something else like 8000.  I added code at one point to make this yser settable, but without that latest version, i think you have to change the code manually and run it interpreted from the base code rather than the distributed version.

Orob-Maslow, 2025-01-17:
It does not run dhcp unless you set up the pi as a hotspot, but that is not part of webcontrol.