WebControlCNC/WebControl #195: Connection Failed or Invalid Firmware

markplow, 2023-01-23:
Trying to get webcontrol set up for the first time.  It seems to connect (shows COM3 in upper right corner) and motors hum on my M2.  However it is giving the error "connection failed or invalid firmware"  I have updated the arduino due to the latest firmware from the makermade site but the error remains.  See attached screen shots.  Any help would be greatly appreciated.

Thank you,
Mark 
[image]
[image]

Orob-Maslow, 2023-01-23:
Webcontrol only works with the mega processor controllers.  The M2 is a Due processor and only works with makerverse.

markplow, 2023-01-24:
Thank you for responding.   I can't get logged into Makerverse so I'm
looking for a work around.....  frustrating.  It would be nice if
Makerverse support responded.

Thanks for your help

Orob-Maslow, 2023-01-24:
There is a nologin workaround for makerverse:

Edit your .makerverse file and after the first open brace ({) .  So it would look like this:
{
"InsecureDangerousGuestAccess":true,

Orob-Maslow, 2023-01-24:
This issue is related to incompatible hardware, not webcontrol.  A workaround was offered.

markplow, 2023-01-24:
> I don't really get why makerverse requires a login...it seems like a
> mistake that makerverse was designed like that...but it's not really a
> WebControl issue :/

Totally agree. Thank you again.

markplow, 2023-01-24:
Thank you!  You are my new favorite person in the world! I owe you a beer
or a coffee!